moonglow/pcan_pro_x #1: Reference hardware

moonglow, 2021-01-15:
### MCU board

* DevEBox STM32F407VGT6 or STM32F407VET6

### CAN transceiver

* TJA1050 module

--------------
You can buy it on TaoBao or AliExpress

moonglow, 2021-01-27:
FAQ:

Q. Do i need to make changes to firmware for STM32F405 ?
A. Nope. You no need to change anything.

Q. Can you provide photo of your MCU test board ?
A. [Yes. As example you can use such boards.](https://github.com/mcauser/MCUDEV_DEVEBOX_F407VET6/blob/master/docs/STM32F407VET6.jpg)

Memfispro, 2021-07-17:
Nice project. What lin driver did you install?

moonglow, 2021-07-17:
@Memfispro Thank you! You can use **TJA1021** as LIN bus driver.

Memfispro, 2021-07-20:
> @Memfispro Thank you! You can use **TJA1021** as LIN bus driver.

Thank you.

rusefillc, 2021-11-10:
Q1; is it theoretically possible to change pinout to match F407 discovery or would it hit any pin/DMA/etc conflicts?

Q2: any chance to have pinout flexible at build time with defines etc?

moonglow, 2021-11-10:
A1: yep you can try absolutely any board you want with F407/F405 IC just connect pins as it described at main page table.
A2: nice tips, i will think about it.

moonglow, 2021-12-27:
Added support for HEX-V2 clone ( it will works as single channel CAN adapter, HW is STM32F415VG based ): [LINK](https://www.aliexpress.com/item/1005003493448473.html)

MC-tecnologias, 2022-11-21:
Hi!
I have a STM32F405 board and want to flash this code.
I dont understand if need bootloadr or not.
I have to program the microcontroller with stlink and that's it?
should I use the bin or the hex?
I installed stm32cubeprogrammer, do you recommend something else?

thank you very much

rusefillc, 2022-11-21:
@MC-tecnologias how much experience do you have with stm32 platform?

Please note that stm32 has build-in bootloader allowing you to program a chip using UART or CAN or USB.

MC-tecnologias, 2022-11-21:
hi @rusefillc !
thank you for replying.
I have experience in microcontrollers but mainly microchip family.
If I plug my board to the PC in DFU mode it recognizes it as "STM32 BOOTLOADER", so I could load the code that way?
should I choose the bin or the hex?

rusefillc, 2022-11-21:
@MC-tecnologias i think you can load either of binary files

MC-tecnologias, 2022-11-21:
On instructions said "Be sure to use PB14/PB15 pins for USB if you wants PRO/PRO FD"
But HW board recommentations has USB on PA11/PA12.
I must flash with PA11/PA12 and then reconect USB on PB14/PB15?

thanks

MC-tecnologias, 2022-11-21:
I have my board programmed!
And windows recognize as "XCAN-USB-PRO CAN Device", but I dont have the driver installed.
Where can I download the driver?

MC-tecnologias, 2022-11-21:
I understand that it no longer appears with the peak vendor-id, but will the software that is waiting for a PCAN-USB recognize it?

moonglow, 2022-11-21:
@MC-tecnologias 
>I understand that it no longer appears with the peak vendor-id, but will the software that is waiting for a PCAN-USB recognize it?

You need to change two number to they original value to make it recognized by your system drivers... You can't load original PCAN drivers without such changes.

MC-tecnologias, 2022-11-22:
Hi, I have my board programmed and recognized by pcan-view and busmaster.
has an 8MHz crystal.
Looking with an oscilloscope nothing comes out of the PB8/PB9 or PB5/PB6 ports when I send a message with any of the software.
Am I missing something?

thanks!

moonglow, 2022-11-23:
@MC-tecnologias Did you attach CAN bus transceivers ( as example TJA1050 ) to that pins ?

MC-tecnologias, 2022-11-23:
Yes, I have a mcp2551.
Anyway, I understand that it is a transciever, so with or without it should come out pulses, correct?

moonglow, 2022-11-23:
If you built correct firmware and connect all as it should be it must works as expected.

p.s: `MCP2551/TJA1050` use `5V` as `VDD`

MC-tecnologias, 2022-11-23:
Yes, I have MCP powered with 5V and STM32 with 3V3.
The board is a drone flight controller with a stm32f405.

MC-tecnologias, 2022-11-23:
On code I only change:
pcan_usb_vendor_id
on pcanpro-usbd.h
And
usbd_manufacturer_string
usbd_product_string_hs
usbd_interface_can_string_hs
On usb_desc.h

It

moonglow, 2022-11-23:
>The board is a drone flight controller with a stm32f405.

Please check your pin schematic, maybe you have pin conflict... maybe you try use PA11/PA12 pins for USB and for not suitable FW for that... i can't imagine all possibles situations.

MC-tecnologias, 2022-11-24:
Hi!
I solder STM32 to a breadboard and connect only it's neccesary pins.
And same result, windows recognize it (I understand that uC <=> PC connection is working) but never send measseges and dont turn on TX led.

moonglow, 2022-11-25:
@MC-tecnologias 
1. You try to solve some abstract problem for my pov, because you do not tell which fw variant you build ( i recommends you 1 channel fd version, or pro fd version ). Please try to understand what we cant saw what you actually have and do
2. Try to debug firmware and set breakpoint on command handler, i do not believe in miracles
3. Send me your source archive with your prebuilt firmware.... and i will try it on my reference dev board

MC-tecnologias, 2022-11-25:
hi @moonglow,
thank you very much for your time and help.
I am using the XCAN PRO version.
What I need is a can-bus interface recognized as PCAN-USB to communicate, through a special software, with a computer via CANOpen protocol.
I tried with the code modified by me but I also tried the V0.4 release without making any changes.
I always get the same result.

kristyandi, 2022-12-14:
> FAQ:
> 
> Q. Do i need to make changes to firmware for STM32F405 ? A. Nope. You no need to change anything.
> 
> Q. Can you provide photo of your MCU test board ? A. Yes. As example you can use such boards. [image]

I found a with the same board and usb connection PA11/12. And I read the F407 manual. If you need a high speed PHY for USB-HS, should you use USB-FS or USB-HS for PB14/15?

moonglow, 2022-12-14:
@kristyandi Hi!

You no need high speed PHY for USB-HS in my case... Readme file refer you which one need to use, **without hi-speed PHY USB-HS will works in USB-FS** mode BUT it has much more endpoints for device such PCAN PRO so it is a little trick to get more endpoints for USB. One channel builds does no need to solder additional pins.

p.s: you can connect PA11 to PB14 and PA12 to PB15 to run any of that firmwares and have ability to run internal STM USB bootloader as example...

P208PUG, 2022-12-22:
> Added support for HEX-V2 clone ( it will works as single channel CAN adapter, HW is STM32F415VG based ): [LINK](https://www.aliexpress.com/item/1005003493448473.html)

flashed this board with st-link and no sign of life...

moonglow, 2022-12-22:
@P208PUG maybe you can provide more information ? which version you flashed in ? how it recognized by your system... did you read about that issue #16  ?

P208PUG, 2022-12-22:
> @P208PUG maybe you can provide more information ? which version you flashed in ? how it recognized by your system... did you read about that issue #16 ? You can also read about that hw [HERE](https://www.rusefi.com/forum/viewtopic.php?f=13&t=2243&p=43737).

so far so good :) proably my file was not good , with the others recognized corectly , red led flashing when connected..

moonglow, 2022-12-22:
>proably my file was not good

In most cases this is main reason. You problem is solved ? That's why i try to ask about more information ^_^

P208PUG, 2022-12-22:
> > proably my file was not good
> 
> In most cases this is main reason. You problem is solved ? That's why i try to ask about more information ^_^

Looks like my problem was compiled sw from fork one , or compiler issue a friend made it for me im only windows user and can't get make to work :)  
my board pinout
[image]

P208PUG, 2023-02-02:
Simple shield 2 can :) will report when arrive
[image]

P208PUG, 2023-02-16:
[image]
[image]

works ok :) just changed usb to solid one :)

yigitcode, 2023-07-20:
Hi when add this project ass makefile project and building it. It makes :
09:31:55 **** Incremental Build of configuration Default for project pcan_prox_01 ****
make all 
make: Makefile:189: pipe: No such file or directory
make: Makefile:202: pipe: No such file or directory
make: Makefile:204: pipe: No such file or directory
make: Makefile:206: pipe: No such file or directory
make BOARD=pro DEBUG=0 OPT=-Os PROTO=Src/pcanpro_protocol.c BOARD_FLAGS='-DPCAN_PRO=1 -DINCLUDE_LIN_INTERFACE=1' elf hex bin
make[1]: Entering directory 'C:/Users/YiÄŸit/Downloads/pcan_pro_x-master (1)/pcan_pro_x-master'
make[1]: Makefile:189: pipe: No such file or directory
make[1]: Makefile:202: pipe: No such file or directory
make[1]: Makefile:204: pipe: No such file or directory
make[1]: Makefile:206: pipe: No such file or directory
make[1]: Nothing to be done for 'elf'.
make[1]: Nothing to be done for 'hex'.
make[1]: Nothing to be done for 'bin'.
make[1]: Leaving directory 'C:/Users/YiÄŸit/Downloads/pcan_pro_x-master (1)/pcan_pro_x-master'
make BOARD=pro_fd DEBUG=0 OPT=-Os PROTO=Src/pcanpro_fd_protocol.c BOARD_FLAGS='-DPCAN_PRO_FD=1 -DINCLUDE_LIN_INTERFACE=1' elf hex bin
make[1]: Entering directory 'C:/Users/YiÄŸit/Downloads/pcan_pro_x-master (1)/pcan_pro_x-master'
make[1]: Makefile:189: pipe: No such file or directory
make[1]: Makefile:202: pipe: No such file or directory
make[1]: Makefile:204: pipe: No such file or directory
make[1]: Makefile:206: pipe: No such file or directory
make[1]: Nothing to be done for 'elf'.
make[1]: Nothing to be done for 'hex'.
make[1]: Nothing to be done for 'bin'.
make[1]: Leaving directory 'C:/Users/YiÄŸit/Downloads/pcan_pro_x-master (1)/pcan_pro_x-master'
make BOARD=fd DEBUG=0 OPT=-Os PROTO=Src/pcanpro_fd_protocol.c BOARD_FLAGS='-DPCAN_FD=1 -DINCLUDE_LIN_INTERFACE=0' elf hex bin
make[1]: Entering directory 'C:/Users/YiÄŸit/Downloads/pcan_pro_x-master (1)/pcan_pro_x-master'
make[1]: Makefile:189: pipe: No such file or directory
make[1]: Makefile:202: pipe: No such file or directory
make[1]: Makefile:204: pipe: No such file or directory
make[1]: Makefile:206: pipe: No such file or directory
make[1]: Nothing to be done for 'elf'.
make[1]: Nothing to be done for 'hex'.
make[1]: Nothing to be done for 'bin'.
make[1]: Leaving directory 'C:/Users/YiÄŸit/Downloads/pcan_pro_x-master (1)/pcan_pro_x-master'

09:31:56 Build Finished. 0 errors, 16 warnings. (took 592ms)
Warnings occures. Then I ignore that warnings and loading hex file pcan_prox_fd hex file in my stm32f4 Discovery evalution board and doesnt excisting any output usb output channel. Am I doing something wrong how should I set up files and. How do I test it ? Is ıt works with directly stm32f4 discovery card ? I thing last is doesnt works because of havent got can transreceiver.

P208PUG, 2024-02-01:
Can you add support to this devboard 3x can-FD https://www.aliexpress.com/item/1005005919904877.html

yigitcode, 2024-02-09:
That board is not mine.

zfhongtao, 2024-03-15:
> [image] [image]
> 
> works ok :) just changed usb to solid one :) can anyone point me where from can get pcanfd .hex

hello bro, where can i buy this vn1610 mainboard?

vido89, 2026-03-01:
> [@mitchdetailed](https://github.com/mitchdetailed) Hi ! I just attach all modules by wires... Special boards is not very suitable for DIY...

Can you show wiring diagram for my [STM32F407VET6](https://www.aliexpress.com/item/1005006761149945.html)  Im looking at [this](https://www.aliexpress.com/item/1005006916261136.html) and [this](https://www.aliexpress.com/item/1005008125932709.html) thanks