moonglow/pcan_pro_x #2: why i cannot setup usbdriver in window10?

cs2132, 2021-01-21:
I  have been used your code ,and has been run.but windows10 cannot set up the usbcan's driver,errcode 31.
[image]

moonglow, 2021-01-21:
Hi,
Please provide me more information about your setup.

1. What hardware board you use.
2. What firmware version you use.

Be aware if your build firmware yourself, you need to use provided HAL libraries it will not works correctly with original HAL lib. Can you check how it works with different PC if it possible. Thank you.

cs2132, 2021-01-24:
how to find firmware version? It should be the latest code。  I used  your provided HAL lib in keil 5 。If I use "pcanpro_protocol.c" file and define PCAN_PRO,  the Window10 can't enumerate. If I use "pcanpro_fd_protocol.c" fileand  define  PCAN_FD, the system can enumerate successfully（“PCAN-USB FD”）. I wonder if there is something wrong with the descriptor of PCAN_PRO.

cs2132, 2021-01-24:
could you give me your email? I'll send you my code to have a look.

moonglow, 2021-01-24:
Ah... you build firmware yourself with Keil. First of all you can try prebuild firmwares, from here: https://github.com/moonglow/pcan_pro_x/releases

Another thing, make sure what you define needed DEFINES while you build your code ( you can check all needed info from Makefile ). As example all PRO versions needs INCLUDE_LIN_INTERFACE. As example for PCAN PRO, you must define

PCAN_PRO=1
INCLUDE_LIN_INTERFACE=1

I try to test all modes before release it, so i think there is no problems. I also will try to retest PCAN PRO firmware to make sure what all is OK. But my build toolchain is slightly different than yours , so i recommend you to use same toolchain or try prebuild binaries for testing.

>>could you give me your email? I'll send you my code to have a look.
you can leave your e-mail here and i will contact to you.

moonglow, 2021-01-25:
I did some tests ( with builds from release page ) all works fine. I think you need to try builds from release page first or more carefully check your own Keil project settings and provided defines.

moonglow, 2021-02-01:
closed for no inactivity.

closed 2021-02-01 by moonglow

JustQY, 2024-04-28:
i have the same issue,and i use f205rc+usb3300,worked on 480Mbit mode,does this mode have problem?

JustQY, 2024-04-28:
oh,i know,you modified the code of usb lib,when i merged, the driver is ok,and the pcan-view shows the device

FrancisChuh, 2024-08-12:
I met the same issue. And I use stm32g4. Build by Keil. My email francischu601@gmail.com. could you help me to have a look? Thank you. @cs2132 @moonglow @JustQY