moonglow/pcan_pro_x #4: LIN protocol implemetation

moonglow, 2021-02-08:
TODO:
LIN protocol implementation stubs

eliocor, 2021-02-08:
Do you want enhance the FW implementation with also LIN implementation?
If yes, which pins will be used?
I ask because I'm thinking to develop  a PCB for the CAN adapter

moonglow, 2021-02-08:
>Do you want to enhance the FW with also LIN implementation?

Yep, I think i will use uart1/uart2 and some pins to do master/slave stuff and so on,it currently in my TODO list only, maybe something be changes in future

eliocor, 2021-02-08:
1) Is there any reference schematics (with LIN interface) compatible with the Peak HW?
2) for CAN FD support, will you ever port the FW to one of this series (STM32G0/STM32G4/STM32H7/STM32L5) which have FD available?
3) if needed I can develop a prototype board using a microcontroller compatible to point 2). The project will be available to everyone

Linjieqiang, 2021-02-08:
STM32G4 is a good choose for CAN FD support.

eliocor, 2021-02-08:
OK for the G4 model. Other than:
- 2x CAN FD
- 2x LIN (with Master mode Enable each one) 
- 1x USB FS
- Tag-connect for debugging purpose
- some LEDs for diagnostics
- EEPROM for settings (or FLASH in EEPROM emulation?

are there some other requirements?

-----------------------
Is there any way I can directly contact you?

moonglow, 2021-02-08:
@eliocor 

> Is there any reference schematics (with LIN interface) compatible with the Peak HW?

Nope, firstly i want to implements all protocol specific thing, hardware realisation it is most easiest part i think, i will add information about it after i finish software part.

> for CAN FD support, will you ever port the FW to one of this series (STM32G0/STM32G4/STM32H7/STM32L5) which have FD available?

I already answered to @Linjieqiang for this question, my main goal is CAN-adapter what can be easy to reproduce and works well with Linux and Windows including professional software. Software support is most important thing for such device, that's why i made this project. About FDCAN part, all protocol related thing already implemented and well tested on real hardware, so where is no problem to port it on G series or any other what support FDCAN, but it possible only for custom solution.

> if needed I can develop a prototype board using a microcontroller compatible to point

It is good idea, but main disadvantage it is more expensive and required more skill to use it ( soldering, components buying, PCB manufacturing and etc )

> Is there any way I can directly contact you?

You can live your e-mail here and i will contact to you, also my profile has public available e-mail too ( i think it must be visible for all registered users )

dreampet, 2021-02-09:
@eliocor 

> * 1x USB FS

Dual-channel CAN FD up to 24 Mbit/s, it needs a USB HiSpeed interface.

eliocor, 2021-02-09:
You are right: microcontroller MUST be in the STM32H7xx line...

moonglow, 2021-02-09:
@dreampet 
> Dual-channel CAN FD up to 24 Mbit/s, it needs a USB HiSpeed interface.

Only in theory, actually FDCAN has slow and fast part of frame + some frame overhead bit + bit stuffing and in real life application you never has non stop bit-stream on 12Mbit speed ( so real speed is slower ), another thing what USB FS 12Mbit is not really 12Mbit too, it is speed of single packet ( MAX is 64byte for FS ) what will transfer from one point to another, and to archive maximum speed we need transfer continuous transfer of data what can't be reached in real application too. Also we have some protocol level overhead...  I think for real life application if you use 1 channel it be OK in most cases on hi speed FDCAN buses too.

p.s: i also read your amobbs forum thread, thanx it makes this project very popular  ^_^

@eliocor 
> You are right: microcontroller MUST be in the STM32H7xx line...

You can use also external HS phy for USB as example USB3300, but you must also use external CAN controllers ( as example mcp1517/18 ) becouse ST can't be configured to use HS USB and two CAN on STM32F series.... you no need to use powerfull MCU to do this, but fast USB is good point for 2/4/6 FDCAN channels and other features

moonglow, 2021-02-17:
Implemented here: [branch](https://github.com/moonglow/pcan_pro_x/tree/pcan_pro_lin), with some issue described here: #5