moonglow/pcan_pro_x #6: USB pins

tomaszduda23, 2021-06-23:
Is it possible to use other pins for USB than PB14/PB15? I would like to use it with STM32F4DISCOVERY which have different pinout.

moonglow, 2021-06-23:
Hi @tomaszduda23 !

>Is it possible to use other pins for USB than PB14/PB15? 

Nope, we can't. We need to use **USB HS** module in **FS** mode ( because we need more endpoints ) and USB HS module mapped to PB14/PB15, you can connect it in parallel with **USB FS** pins. And you will use USB HS module and vise versa. I assume you use [that board](https://www.st.com/en/evaluation-tools/stm32f4discovery.html) and that [board schematic](https://www.st.com/content/ccc/resource/technical/layouts_and_diagrams/schematic_pack/group1/0f/91/8b/39/b3/78/4d/c4/MB997-F407VGT6-B02_Schematic/files/MB997-F407VGT6-B02_Schematic.pdf/jcr:content/translations/en.MB997-F407VGT6-B02_Schematic.pdf), so you need connect PB14 ( USB DM ) to PA11 ( you can use R58 as solder point ) and PB15 ( USB DP ) to PA12  ( you can use R60 as solder point ).

tomaszduda23, 2021-06-23:
I got it. USB works with mandatory ports. Thanks.

closed 2021-06-23 by tomaszduda23