moonglow/pcan_pro_x #8: can't open device

tomaszduda23, 2021-06-27:
For some reason it does not work with mainline driver. After getting pcan driver it is detected correctly but works only sometime. After building pcan driver with #define DEBUG_TRACE and DBG=DEBUG works every single time. It also does not work on win32. Any clue?

`[ 4224.616141] pcan: data [brp=0 tseg1=0 tseg2=0 sjw=0 sp=0]`
`[ 4224.616144] pcan: dumping sent message (12 bytes):`
`[ 4224.616146] pcan: 01 00 00 00 02 00 01 00 06 00 1c 00`
`[ 4224.616251] pcan: dumping sent message (8 bytes):`
`[ 4224.616253] pcan: 01 00 00 00 05 00 00 00`
`[ 4227.613400] pcan: pcan_usbpro_cmd_send_ex(pcan32): failed to send sync cmd (err -2)`
`[ 4227.613402] pcan: can't open device hardware itself (err -2)!`
`[ 4227.613530] pcan: PCAN-USB Pro CAN2: 500 items Tx FIFO allocated`
`[ 4227.613537] pcan: PCAN-USB Pro CAN2: 500 items Rx FIFO allocated`
`[ 4227.613537] pcan: pcan_usb_stop(CAN2), minor=33`
`[ 4227.613538] pcan: PCAN-USB Pro CAN2 rx=0 tx=0`
`[ 4227.613539] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(500): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4227.613540] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(100): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4227.613541] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(140): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4227.613541] pcan: opening PCAN-USB Pro CAN2 with nominal bittiming:`
`[ 4227.613541] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(562): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4227.613542] pcan: PCAN-USB Pro(CAN2): time=1624804228.464147s: opening with clk=56000000 bitrate=500000 dbitrate=2000000 (flags=00000000h)`
`[ 4227.613543] pcan: nominal [brp=7 tseg1=13 tseg2=2 sjw=1 sp=8750]`
`[ 4227.613543] pcan: data [brp=0 tseg1=0 tseg2=0 sjw=0 sp=0]`
`[ 4227.613544] pcan: dumping sent message (12 bytes):`
`[ 4227.613545] pcan: 01 00 00 00 02 01 00 00 06 00 1c 00`
`[ 4230.613466] pcan: pcan_usbpro_cmd_send_ex(pcan33): failed to send sync cmd (err -2)`
`[ 4230.613468] pcan: can't open device hardware itself (err -2)!`
`[ 4230.615491] pcan: PCAN-USB Pro CAN1: 500 items Tx FIFO allocated`
`[ 4230.615498] pcan: PCAN-USB Pro CAN1: 500 items Rx FIFO allocated`
`[ 4230.615499] pcan: pcan_usb_stop(CAN1), minor=32`
`[ 4230.615500] pcan: PCAN-USB Pro CAN1 rx=0 tx=0`
`[ 4230.615501] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(500): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4230.615502] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(100): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4230.615503] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(140): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4230.615503] pcan: opening PCAN-USB Pro CAN1 with nominal bittiming:`
`[ 4230.615503] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(562): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4230.615504] pcan: PCAN-USB Pro(CAN1): time=1624804231.466128s: opening with clk=56000000 bitrate=500000 dbitrate=2000000 (flags=00000000h)`
`[ 4230.615505] pcan: nominal [brp=7 tseg1=13 tseg2=2 sjw=1 sp=8750]`
`[ 4230.615505] pcan: data [brp=0 tseg1=0 tseg2=0 sjw=0 sp=0]`
`[ 4230.615507] pcan: dumping sent message (12 bytes):`
`[ 4230.615508] pcan: 01 00 00 00 02 00 00 00 06 00 1c 00`
`[ 4233.613402] pcan: pcan_usbpro_cmd_send_ex(pcan32): failed to send sync cmd (err -2)`
`[ 4233.613404] pcan: can't open device hardware itself (err -2)!`
`[ 4243.876776] [UFW BLOCK] IN=wlo1 OUT= MAC=bc:17:b8:cc:f0:33:d0:c6:37:e9:b3:17:08:00 SRC=192.168.0.204 DST=192.168.0.196 LEN=56 TOS=0x00 PREC=0x00 TTL=128 ID=10367 PROTO=UDP SPT=62974 DPT=2054 LEN=36 `
`[ 4246.408709] pcan: PCAN-USB Pro CAN1: 500 items Tx FIFO allocated`
`[ 4246.408727] pcan: PCAN-USB Pro CAN1: 500 items Rx FIFO allocated`
`[ 4246.408730] pcan: pcan_usb_stop(CAN1), minor=32`
`[ 4246.408731] pcan: PCAN-USB Pro CAN1 rx=0 tx=0`
`[ 4246.408735] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(500): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4246.408737] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(100): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4246.408739] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(140): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4246.408740] pcan: opening PCAN-USB Pro CAN1 with nominal bittiming:`
`[ 4246.408742] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(562): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4246.408745] pcan: PCAN-USB Pro(CAN1): time=1624804247.259463s: opening with clk=56000000 bitrate=500000 dbitrate=2000000 (flags=00000000h)`
`[ 4246.408747] pcan: nominal [brp=7 tseg1=13 tseg2=2 sjw=1 sp=8750]`
`[ 4246.408748] pcan: data [brp=0 tseg1=0 tseg2=0 sjw=0 sp=0]`
`[ 4246.408751] pcan: dumping sent message (12 bytes):`
`[ 4246.408753] pcan: 01 00 00 00 02 00 00 00 06 00 1c 00`
`[ 4249.405424] pcan: pcan_usbpro_cmd_send_ex(pcan32): failed to send sync cmd (err -2)`
`[ 4249.405425] pcan: can't open device hardware itself (err -2)!`
`[ 4249.405509] pcan: PCAN-USB Pro CAN2: 500 items Tx FIFO allocated`
`[ 4249.405516] pcan: PCAN-USB Pro CAN2: 500 items Rx FIFO allocated`
`[ 4249.405517] pcan: pcan_usb_stop(CAN2), minor=33`
`[ 4249.405518] pcan: PCAN-USB Pro CAN2 rx=0 tx=0`
`[ 4249.405519] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(500): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4249.405520] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(100): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4249.405520] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(140): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4249.405521] pcan: opening PCAN-USB Pro CAN2 with nominal bittiming:`
`[ 4249.405521] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(562): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4249.405522] pcan: PCAN-USB Pro(CAN2): time=1624804250.256260s: opening with clk=56000000 bitrate=500000 dbitrate=2000000 (flags=00000000h)`
`[ 4249.405522] pcan: nominal [brp=7 tseg1=13 tseg2=2 sjw=1 sp=8750]`
`[ 4249.405523] pcan: data [brp=0 tseg1=0 tseg2=0 sjw=0 sp=0]`
`[ 4249.405524] pcan: dumping sent message (12 bytes):`
`[ 4249.405525] pcan: 01 00 00 00 02 01 00 00 06 00 1c 00`
`[ 4252.405329] pcan: pcan_usbpro_cmd_send_ex(pcan33): failed to send sync cmd (err -2)`
`[ 4252.405331] pcan: can't open device hardware itself (err -2)!`
`[ 4253.536224] pcan: PCAN-USB Pro CAN1: 500 items Tx FIFO allocated`
`[ 4253.536243] pcan: PCAN-USB Pro CAN1: 500 items Rx FIFO allocated`
`[ 4253.536245] pcan: pcan_usb_stop(CAN1), minor=32`
`[ 4253.536247] pcan: PCAN-USB Pro CAN1 rx=0 tx=0`
`[ 4253.536251] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(500): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4253.536253] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(100): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4253.536255] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(140): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4253.536256] pcan: opening PCAN-USB Pro CAN1 with nominal bittiming:`
`[ 4253.536258] pcan: /var/lib/dkms/peak-linux-driver/8.12.0/build/src/pcanfd_core.c(562): clk=56MHz btr=500kbps sp=8750 [BRP=7 TSEGx=13/2 SJW=1 TQ=125]`
`[ 4253.536261] pcan: PCAN-USB Pro(CAN1): time=1624804254.387022s: opening with clk=56000000 bitrate=500000 dbitrate=2000000 (flags=00000000h)`
`[ 4253.536263] pcan: nominal [brp=7 tseg1=13 tseg2=2 sjw=1 sp=8750]`
`[ 4253.536264] pcan: data [brp=0 tseg1=0 tseg2=0 sjw=0 sp=0]`
`[ 4253.536267] pcan: dumping sent message (12 bytes):`
`[ 4253.536269] pcan: 01 00 00 00 02 00 00 00 06 00 1c 00`
`[ 4256.533372] pcan: pcan_usbpro_cmd_send_ex(pcan32): failed to send sync cmd (err -2)`
`[ 4256.533373] pcan: can't open device hardware itself (err -2)!`

moonglow, 2021-06-27:
@tomaszduda23 
>> It also does not work on win32. Any clue?

It works perfect for me under Linux and Windows, maybe you can provide more information ? Did you build firmware yourself or you try precompiled binaries ? I recommends to use precompiled binaries first.

tomaszduda23, 2021-06-27:
I built myself first. Now I just tried precompiled pcan_pro_hw_2021_06_11.hex it still does not work. Which driver do you used on linux?

I noticed that after connecting device first command seems to be ok. Than each next command fails:

`pcan: pcan_usbpro_set_baudrate(PCAN-USB Pro CAN1, baudrate=001c0006h)`
`pcan: pcan_usbpro_cmd_send_ex(pcan32, flags=1): ->EP#01`
`pcan: dumping sent message (12 bytes):`
`pcan: 01 00 00 00 02 00 01 00 06 00 1c 00`     <= set_baudrate seems to work
`pcan: pcan_usbpro_submit_cmd_complete(pcan32) = 0`
`pcan: pcan_usbpro_set_silent_mode(PCAN-USB Pro CAN1, silent_on=0)`
`pcan: pcan_usbpro_cmd_send_ex(pcan32, flags=1): ->EP#01`
`pcan: dumping sent message (8 bytes):`
`pcan: 01 00 00 00 05 00 00 00 `    <= than each command fails with error
`pcan: pcan_usbpro_submit_cmd_complete(pcan32) = -2`
`pcan: pcan_usbpro_submit_cmd_complete(pcan32) = -2`
`pcan: pcan_usbpro_submit_cmd_complete(pcan32) = -2`
`pcan: pcan_usbpro_cmd_send_ex(pcan32): failed to send sync cmd (err -2)`
`pcan: pcan_usb_free_irq(pcan32)`
`pcan: can't open device hardware itself (err -2)!`

Maybe it is issue with my hardware itself? Strange thing is that it works randomly. I can provide more information if you need.

`$ cat /proc/pcan`
`*------------- PEAK-System CAN interfaces (www.peak-system.com) -------------`
`*------------- Release_20210505_n (8.12.0) Jun 27 2021 17:16:58 --------------`
`*---------------------------- [dbg] [mod] [usb] -----------------------------`
`*--------------------- 2 interfaces @ major 510 found -----------------------`
`*n -type- -ndev- --base-- irq --btr- --read-- --write- --irqs-- -errors- status`
`32    usb   -NA-    30d5e 255 0x001c 00000000 00000000 00000000 00000000 0x0000`
`33    usb   -NA-    30d5e 255 0x001c 00000000 00000000 00000000 00000000 0x0000`

moonglow, 2021-06-27:
please check updated information about pin conflicts what i provided, but it works for any Linux distro i use, Ubuntu 18.04, Ubuntu 21.04, Windows 7 x86, Windows 10 x64 and not only for me. i think this is hardware issue

moonglow, 2021-06-27:
Another thing, you can't pins left open and connected to audiochip, i think I'm sure what this is main problem because i have a lot of users ( i also use it myself ) has no problem. You can't left CAN pins open because it will never get IDLE state while configuration, so what i will do in your situation, first of all you need to resolve all pin conflicts and you need also add CAN phy or if you haven't you need to add something like that to CAN init pins:

```
#define CAN1_RX  B, 8, MODE_AF_OD, PULLUP, SPEED_FREQ_VERY_HIGH, AF9_CAN1
#define CAN1_TX  B, 9, MODE_AF_PP, NOPULL, SPEED_FREQ_VERY_HIGH, AF9_CAN1

#define CAN2_RX  B, 5, MODE_AF_OD, PULLUP, SPEED_FREQ_VERY_HIGH, AF9_CAN2
#define CAN2_TX  B, 6, MODE_AF_PP, NOPULL, SPEED_FREQ_VERY_HIGH, AF9_CAN2
```

Another solution it is remove all CAN related function ) and you will get just working USB stub without actual CAN functionality

tomaszduda23, 2021-06-27:
I'm using SN65HVD230D. As long as CAN side is "floating" it always crash like bellow. It also crashes in the car.

`(gdb) bt`
`#0  _exit (rc=rc@entry=1) at ../../../../../../../libgloss/libnosys/_exit.c:17`
`#1  0x08000db6 in abort () at ../../../../../../../../newlib/libc/stdlib/abort.c:59`
`#2  0x0800023e in __assert_func (file=0x800690c "Src/pcanpro_can.c", line=<optimized out>, func=<optimized out>, failedexpr=0x8006a34 "0") at ../../../../../../../../newlib/libc/stdlib/assert.c:62`
`#3  0x0800265e in pcan_can_set_bitrate_ex ()`
`#4  0x080063f0 in pcan_protocol_set_baudrate ()`
`#5  0x08006568 in pcan_protocol_process_data ()`
`#6  0x08002f7a in device_data_out ()`
`#7  0x08003786 in HAL_PCD_IRQHandler ()`
`#8  0x08002172 in main ()`

Current CAN configuration.
`#define CAN2_RX  B, 5, MODE_AF_PP, NOPULL, SPEED_FREQ_VERY_HIGH, AF9_CAN2`
`#define CAN2_TX  B, 7, MODE_AF_PP, NOPULL, SPEED_FREQ_VERY_HIGH, AF9_CAN2`

`#define CAN1_RX  B, 8, MODE_AF_OD, PULLUP, SPEED_FREQ_VERY_HIGH, AF9_CAN1`
`#define CAN1_TX  B, 11, MODE_AF_PP, NOPULL, SPEED_FREQ_VERY_HIGH, AF9_CAN1`

moonglow, 2021-06-28:
>  it always crash like bellow.
>  return HAL_ERROR; < - this one fails during pcan_can_init_ex

because CAN periphery can't get idle state, whats why i recommends before to configure it as AF_OD with PULLUP useful if you have board without CAN transceivers. If CAN periphery can't be init it goes to assert( 0 ) construction, because it is fatal error. 

**About this:**
```
#define CAN2_RX B, 5, MODE_AF_PP, NOPULL, SPEED_FREQ_VERY_HIGH, AF9_CAN2
#define CAN2_TX B, 7, MODE_AF_PP, NOPULL, SPEED_FREQ_VERY_HIGH, AF9_CAN2

#define CAN1_RX B, 8, MODE_AF_OD, PULLUP, SPEED_FREQ_VERY_HIGH, AF9_CAN1
#define CAN1_TX B, 11, MODE_AF_PP, NOPULL, SPEED_FREQ_VERY_HIGH, AF9_CAN1
```

You **can't use** CAN peripheral with any pins you want. As example in your case ( according to device [datasheet](https://www.st.com/resource/en/datasheet/stm32f407vg.pdf), Table 7 ) :

**PB5** - has alternate function as **CAN2_RX**
**PB7** - does not have alternate function for CAN periphery.
**PB8** - has alternate function as **CAN1_RX**
**PB11** -  does not have alternate function for CAN periphery.

So... i think this is your main problem you can try this _TEST_ firmware without CAN pins used ( i just make it virtual ), just to check if you solder your USB pins correct, if it will works for you, next step you will need to sure what pinout is correct and i think all be OK.

[pcan_pro_virt_can.zip](https://github.com/moonglow/pcan_pro_x/files/6724127/pcan_pro_virt_can.zip)

tomaszduda23, 2021-06-30:
You are right. I forgot about alternate functions table. Bellow works correctly.

`#define CAN1_RX  B, 8, MODE_AF_PP, NOPULL, SPEED_FREQ_VERY_HIGH, AF9_CAN1`
`#define CAN1_TX  D, 1, MODE_AF_PP, NOPULL, SPEED_FREQ_VERY_HIGH, AF9_CAN1`

closed 2021-06-30 by tomaszduda23