moonglow/pcan_pro_x #11: Difference with pcan_cantact

jxltom, 2021-07-09:
It looks like fd frames are also not supported in this firmware.

So what is the difference with this firmware https://github.com/moonglow/pcan_cantact? The latter one also supports pcan-view.

moonglow, 2021-07-09:
@jxltom 
Hi !
Some answer was already given here #4 , but i will try to explain more.

So firmware is really supports **CANFD** frames, all needed protocol handling and so on are implemented and well tested. Hardware ( **STM32F4xx** ) physically can't send/receive FD frame due **bxCAN** does not support them. PCAN CANtact is **1 channel** basic CAN adapter it was made as alternative firmware for very popular CAN hardware ( CANable, CANtact and Entreé ), PCAN PRO is **2 channel** CAN + 2 channel LIN adapter, PCAN PRO FD - same as PRO but supports **FDCAN** extension. All of them uses **three different USB level protocol implementation** so it was a cool thing to implement them all. Another thing, i select my base hardware as most cheapest possible, to be sure what many peoples can DIY your own CAN adapter. Anyone can add CANFD to current implementation as example by creating custom `pcanpro_can.c` and connect something like **MCP2517** via SPI to board or port my firmware to something what already has builtin **FDCAN** hardware such STM32G4xx series.

jxltom, 2021-07-09:
@moonglow Thanks a lot for the explanation!

closed 2021-07-09 by jxltom