moonglow/pcan_pro_x #12: Not an issue

PKilgannon, 2021-07-11:
Has anybody ported this awesome software to the STM32H750? 

If I'm not mistaken it has 9-endpoints+0 and two canfd channels. This could solve the second LIN channel problem and add the canfd channels. Dev boards are easily got from Ebay and Aliexpress.

BTW, works great on STM32F429!

[image]

rusefillc, 2021-11-10:
@PKilgannon what is this board on the picture? :)

PKilgannon, 2021-11-10:
@rusefillc 

It's a pcb I made to test out moonglows firmware. I tested with an stm32f429 but its pin compatible with the stm32H7 series. The hardware supports two channels of CAN/LIN individually isolated. You can also add a orangepi zero sbc to play with ;)
I haven't got any stm32H7 to test due to pandemic shortage. Hopefully I can get my grubby hands on some next year.

rusefillc, 2021-11-10:
@PKilgannon that this is KiCAD? :) orange pi is what caught my attention I've played with those as well.

even with pandemic it's still possible to grab some h7 here and there, is that 144 or 176 pin?

PKilgannon, 2021-11-10:
@rusefillc it's designed with Altium, 144pin package, they're usually pin compatible across families. 
I refuse to pay 10x the price for parts on a none commercial project :p

rusefillc, 2021-11-10:
@PKilgannon newak promises to have STM32H743ZIT6 in two weeks. I've purchased STM32H743ZIT6 from them in Sept and they did in fact ship 10 days later. H7 could be different factory for ST thus different situation.

PKilgannon, 2021-11-12:
@rusefillc I'm traveling until next year but I'll check what the situation is like in Europe again I. January. Thanks for the heads Up.

FrancisChuh, 2024-08-07:
@PKilgannon any update about support CANFD on stm32h7?
@moonglow the pc couldn't recognize the device, could you give me some support? Thank you