moonglow/pcan_pro_x #13: STM32f407 Standard

uControllerLab, 2021-11-04:
Standard usb pin on most devebox stm32f407 is 

PA11 USB_DM
PA12 USB_DP

Can you please make CubeMX project to easy  change pin configuration

Thanks

moonglow, 2021-11-04:
@uControllerLab Hi !

>Standard usb pin on most devebox stm32f407 is

I already explain here about USB, i need more USB endpoints that's why i use **USB HS in FS mode** and this peripheral located on another pins.

>Can you please make CubeMX project to easy change pin configuration

Sorry but it is useless i do not use CubeMX peripheral config for most already made peripheral.

uControllerLab, 2021-11-05:
Thank you for the explanation!

it work with devebox black stm32f407 when make bridge
PA11->PB14
PA12->PB15

closed 2021-11-05 by moonglow