moonglow/pcan_pro_x #16: !ACHTUNG! PEAK company copyrighted data

moonglow, 2022-03-10:
Due PEAK company request to remove copyrighted data from USB descriptor, i removed such data from my repo. Sorry but we must respect other companies.

Q: Firmware will works ?
A: By default firmware not works anymore with PEAK drivers. USD VID set to another value.

zhangjingliu, 2025-01-05:
Hi, bro, I know how to change it to work well, the USB VID and Vendor strings, by the way, Does this firmware can support CANFD? and can use it for bootloader ota ,such as UDS flash bootloader, it can use for this ???

wchf204, 2025-07-16:
> 嗨，兄弟，我知道如何更改它才能正常工作，USB VID 和供应商字符串，顺便问一下，这个固件可以支持 CANFD 吗？并且可以将其用于引导加载程序 ota，例如 UDS flash 引导加载程序，它可以用于这个吗？

how to change,thks