moonglow/pcan_pro_x #18: Passthru support

c0d3z3r0, 2023-01-12:
Have you ever tried to use this with the J2534 Passthru API? I got a VCDS clone working with this, using PCAN-OBD-2, so the adapter itself works. However, at least ODIS says that communication with the interface does not work.

moonglow, 2023-01-12:
I already heard about that issue, but PEAK CEO ask me to completely stop all my development for PCAN protocol compatible device, because there are very unhappy this.

c0d3z3r0, 2023-01-12:
I see. Feel free to contact me on IRC libera.chat for some exchange :)

c0d3z3r0, 2023-01-17:
It might be worth looking at Sardine CAN (a open source J2534 can interface), which also has a open-source driver for windows.
This would allow this project to continue without copyright problems.

- https://github.com/hackingvolvo/SardineCAN-Arduino
- https://github.com/hackingvolvo/SardineCAN-Win32

moonglow, 2023-01-18:
> This would allow this project to continue without copyright problems.

Looks like it already done and you can use it with their recommended hardware ?

zhangjingliu, 2025-01-18:
> PEAK CEO

PEAK CEO will die, go to the hell, he is a baster